umrover/embedded-testbench #163: explore cameras

tabiosg, 2022-09-20:
yes

sashreek1, 2022-09-23:
Here's the new ROS camera code (untested):
https://github.com/tabiosg/mrover-ros/tree/cameras_ros

Camera code before switching to ROS:
https://github.com/umrover/mrover-workspace/tree/main/jetson/cameras

tabiosg, 2022-09-23:
@sashreek1 wait are u telling them to create a program using jetson-utils?

tabiosg, 2022-09-23:
imo it would be useful if they listed all the options for camera programs instead

sashreek1, 2022-09-26:
Yeah agreed. Try looking into gstreamer or OpenCV for this. These are just options try looking at other options too.

qhdwight, 2022-09-26:
jetson-utils uses gstreamer. What is wrong with it? It will actually fit perfectly into our code base since it uses CMake

qhdwight, 2022-09-26:
https://github.com/dusty-nv/jetson-utils/blob/35593c5ad2c1c62f6f0166d6581945fb58fd1f7b/camera/v4l2Camera.cpp#L86

There is a timeout parameter. This should figure the freezing issue.

tabiosg, 2022-09-26:
@sashreek1 Ok I talked with Quintin and we should stick with jetson-utils for now. We can look at other alternatives too but for now just write this program