umrover/embedded-testbench #166: Implement servo.c for science nucleo

tabiosg, 2022-09-22:
Use this as a reference
https://github.com/umrover/embedded-testbench/blob/guts/science-dev/science-dev/Core/Inc/servo.h

tabiosg, 2022-09-23:
https://github.com/umrover/embedded-testbench/blob/rover-22/science/science_nucleo_g0/Core/Src/servo.c

tabiosg, 2022-09-23:
Check out of guts/science-dev and make a new branch called something like eric/servo or wis/servo. Create the servo.c file in Core/Src folder. Use [this ](https://github.com/umrover/embedded-testbench/blob/rover-22/science/science_nucleo_g0/Core/Src/servo.c) code as reference for working code. Feel free to change up the header file if needed

sashreek1, 2022-09-23:
Here's the old science servo code:
https://github.com/umrover/embedded-testbench/blob/rover-22/science/science_nucleo_g0/Core/Src/servo.c

sashreek1, 2022-09-23:
PWM signal resources :
https://deepbluembedded.com/stm32-pwm-example-timer-pwm-mode-tutorial/

tabiosg, 2022-10-08:
Closed by #175

PR #175 (2022-10-01): Add basic servo.c implementation
Closes #166 

Tested with [SG92R](http://www.ee.ic.ac.uk/pcheung/teaching/DE1_EE/stores/sg90_datasheet.pdf)

Test in `main.c` runs on Timer 2 Channel 1

$F_{PWM} = \frac{F_{CLK} }{(ARR + 1) \times (PSC + 1)}$

Clock Constant: $F_{CLK} = 16 MHz$
Servo Frequency: $F_{PWM} = 50 Hz$ (Specific to servo, see datasheet)
$PSC = 1599$
$ARR = 199$

**NOTE** 
The servo is limited to ~ $120^\circ$ range
closed 2022-10-08 by tabiosg